jhavl/robotacademy-code #4: MathWorks Issues *In progress, will continue updating*

Tjeard0, 2023-07-23:
Major:

Minor:

**Analyzing a very simple 1- joint robot arm, Analyzing a very simple 2-joint planar robot arm, Analyzing a very simple 3-joint planar robot arm, Analyzing a robot arm that moves in 3D:**

**Analyzing a very simple 1- joint robot arm, Analyzing a very simple 2-joint planar robot arm, Analyzing a very simple 3-joint planar robot arm, danavit-hartenberg notation, Redundant robots, Velocity Ellipse in 2D, Jacobian and number of robot joints, Forces acting on robot links, Rigid body dynamics, Gravity and payload, inertia and acceleration coupling, velocity coupling:** No simple method for generating the planar arms or the 20 joint arm, this could be written up as a separate library function however using the rigid body tree modeling method for MathWorks. 

**Multi-dimensional Trajectory:** when using the equivalent function `contopptraj([first,last],vellim, acclim);` to create the constrained trajectories, the constraints for MathWorks and robotics toolbox aren't the same, robotics toolbox allows for specifying start and end velocity while mathworks has to specify velocity and acceleration limits (unable to set start and end velocity, assumed to be 0). This produces different graphs

petercorke, 2023-07-25:
@Tjeard0, it'd be easier if these were individual issues.  For each one please include the name of the folder on GH that it refers to.  These are all good points so please do this and we can then have a threaded discussion.

More generally:
* if a capability doesn't exist, eg. no symbolic variables in kinematics, just put that in a comment in the file
* please use `*` not `.*` to multiply MATLAB matrices and pose objects
* `se2(so2(q1, "theta"))` can be written much more concisely as `se2(q1, "theta")` 
* it's Denavit not Danavit

Tjeard0, 2023-08-07:
@petercorke I have added in the new issues for things that should be fixable. The rest of the things in this issue will be written up as comments in the files as it is relating to missing functionality.

closed 2024-01-08 by Tjeard0