jhavl/robotacademy-code #14: Images from videos code is convoluted in python

Tobias-Fischer, 2023-09-09:
https://robotacademy.net.au/masterclass/getting-images-into-a-computer/?lesson=632

@Tjeard0 is there really no easier way to extract images?

Tjeard0, 2023-09-09:
This definitely isn't the way the code was meant to be used looking at the documentation, I set it up to be as close to the matlab implementation as possible.

Currently it is set up using iterators so normally this would be processed in a for loop which is what is described in the documentation:

`from machinevisiontoolbox import VideoFile
video = VideoFile("traffic_sequence.mpg")
len(video)
for im in video:
  # process image`

I have checked and unfortunately the .grab function (used in the matlab video ) only exists for video cameras and not video files in the machine vision toolbox.

I can re-write the file if you like to use a for loop instead which would be the intended for the python implementation.

Tjeard0, 2023-09-09:
In the same folder I have written up a file: python-simplified.py which uses a for loop instead and limits to 6 frames like the matlab implementation. Let me know if this is a nicer way of doing it and I can update the main file

closed 2023-11-14 by Tjeard0