jhavl/robotacademy-code #15: Double is float64, not float32

Tobias-Fischer, 2023-09-09:
@Tjeard0 could you please go through the python codes and check that we use float64 when Peter uses double in MATLAB? This could be confusing. E.g. https://robotacademy.net.au/masterclass/getting-images-into-a-computer/?lesson=633

Tjeard0, 2023-09-09:
Ah im very sorry I will go update that and close the comment when I am finished!

Tjeard0, 2023-09-09:
The following error occurs when I try and use float64 instead of float32 in the video: template-matching

cv2.error: OpenCV(4.7.0) D:\a\opencv-python\opencv-python\opencv\modules\imgproc\src\templmatch.cpp:1164: error: (-215:Assertion failed) (depth == CV_8U || depth == CV_32F) && type == _templ.type() && _img.dims() <= 2 in function 'cv::matchTemplate'

All other videos using float32 have been converted to float64

closed 2023-11-14 by Tjeard0