jhavl/robotacademy-code #19: Region features video should draw ellipse, not circle

Tobias-Fischer, 2023-09-12:
https://robotacademy.net.au/masterclass/feature-extraction/?lesson=695

Tobias-Fischer, 2023-09-12:
It's defined in: https://github.com/bdaiinstitute/spatialmath-python/blob/07c6f62460030fd2c4f45112611e783abfbd8e4f/spatialmath/base/graphics.py#L820

Tjeard0, 2023-09-12:
Ah I was having issues with the plot function earlier which is why I was
using the circle, I'll try again tonight to get it up and running correctly!

Tobias-Fischer, 2023-09-12:
See also https://github.com/petercorke/RVC3-python/blob/main/notebooks/chap12.ipynb (Figure 12.11 in the book)

Tjeard0, 2023-09-12:
Ah perfect, I had not seen that before so it will definitely help, thank
you!

Tjeard0, 2023-09-14:
This should be added in now with the correct functionality, I had some issues with it not displaying correctly on my system (tested both in a normal python and jupyter notebook) but it was working in the python notebook on google colab and I have made sure that the ellipses are being plotted correctly corresponding to the sharks in the image.

closed 2023-09-14 by Tjeard0