jhavl/robotacademy-code #22: Many videos do not contain a code session, but still references to code

Tobias-Fischer, 2023-09-14:
Ideally we should add simple code snippets to those as well, see e.g. https://robotacademy.net.au/masterclass/spatial-operators/?lesson=678 - this way it's easy to see which function to use in other toolboxes.

Tjeard0, 2023-09-14:
Ah I'm very sorry, I had written these down to add code in for them but had completely forgotten about it. There should be 7 videos in total being:

Planar Homography
Separating Intensity and Color
Mathematical Morphology
Rank Filtering
kernels
Inverse Tangent function
Summary of digital images

I will make sure to complete those soon!

Tobias-Fischer, 2023-09-14:
Please also note that you currently push to the wrong (non main) branch.

Tjeard0, 2023-09-14:
Yeah I still have been receiving errors when trying to use the main branch.
I can try again tonight and I'll link the error I get.

Tjeard0, 2023-09-14:
This is the issue I keep running into when attempting to upload directly to the main branch. 

[image]

Tjeard0, 2023-10-15:
Sorry for the large delay in me getting these done, I have been moving houses and it has been taking up significantly more time than I had wanted. 

All of these videos should now have code snippets for rvc1, rvc2, mathworks when applicable and python

closed 2023-10-15 by Tjeard0