jhavl/robotacademy-code #24: Colour images has code but shows as not having code

Tobias-Fischer, 2023-09-20:
https://robotacademy.net.au/masterclass/color/?lesson=716

Tjeard0, 2023-10-03:
I have taken a look and the github repo should have code in the folder for this video, there is a discrepency between the name of the folder and the name of the video however as they do not match on the lesson website as well. 

Lesson name: Color Images in MATLAB
Youtube Video name: Color Image in MATLAB

I am not sure if this is related to how the code on the website is drawing in from the github repo and that the missing code is due to the missmatch in the naming between the video and the actual lesson/folder

jhavl, 2023-12-06:
Fixed, was a name mismatch as guessed
Thanks!

closed 2023-12-06 by jhavl