jhavl/robotacademy-code #31: Execution of script closest.m errors

Tjeard0, 2024-01-08:
when using the code

`m = f1.match(f2);`

in the finding correspondence using surf features video the following error is produced

Execution of script closest as a function is not supported:
C:\Users\*USERNAME*\AppData\Roaming\MathWorks\MATLAB Add-Ons\Toolboxes\Machine Vision Toolbox for MATLAB\vision\closest.m

Error in SurfPointFeature/match (line 122)
            [cor2,err] = closest(D1, D2);

Error in rvc2 (line 18)
m = f1.match(f2);

This same issue for closest.m occurs in the advanced mathematical morphology code as well for the colorkmeans function

I have linked a text file for the contents of the closest.m file to help with this issue. If you follow the file directories shown earlier with your own username this should point to the matlab file as well. 

[closest.m.txt](https://github.com/jhavl/robotacademy-code/files/13867009/closest.m.txt)

Tobias-Fischer, 2024-01-08:
It seems like this is the error reported here, plus suggested fix: https://github.com/petercorke/machinevision-toolbox-matlab/blob/master/mex/README

@Tjeard0 can you please check?

Tjeard0, 2024-01-08:
From the looks of it that should fix the issue, I'll test this when I get back home and update the files and GitHub issue respectively if it works

Tobias-Fischer, 2024-01-08:
If it works, could you please add a link to this README in the relevant video scripts?

Tjeard0, 2024-01-08:
Sounds good, I'll make sure to add in the link!

Tjeard0, 2024-01-19:
After testing the make command as recommende in the readme all of the mex files compile succesfully but the apriltags.c file fails. It links to a website: https://april.eecs.umich.edu/wiki/index.php/AprilTags

which produces an error 404 when I try and access it. After the make command I also tried to run the file and it was still producing the error

Tobias-Fischer, 2024-02-29:
I didn't have any issues compiling this, closing here.

closed 2024-02-29 by Tobias-Fischer